robotics-in-concert/rocon_multimaster #295: UDP communication between different ROS networks

garaemon, 2015-01-07:
Is it possible to communicate different ROS networks on UDP by using rocon?

jihoonl, 2015-01-08:
rocon is just setting up channels among multiple ros masters. The actual communication is via ros communication layer. So UDP should be possible as roscpp supports it.

garaemon, 2015-01-08:
Does this document describe how rocon works?
http://wiki.ros.org/rocon_gateway_tutorials/Tutorials/indigo/Flipping

stonier, 2015-01-08:
That just describes one method of getting ros communications from one master to another master.

Rocon uses this tool, but there are many other concepts in Rocon too.

stonier, 2015-02-27:
Just for reference, the answer here was yes...roscpp udp is certainly possible. To flip python topics (no udp in rospy), just drop in the ros relays and flip that.

Closing this :)

closed 2015-02-27 by stonier