robotics-in-concert/rocon_multimaster #296: [rocon_gateway] expose default.yaml as roslaunch args

jihoonl, 2015-04-10:
So user can easily set their own.

jihoonl, 2015-04-10:
@stonier,
This is nice idea to let users to configure gateway easily. However, it also creates conflict between `default.yaml` and roslaunch args.

For example, `gateway_name` is configured by `default.yaml` first and then overridden by roslaunch arg. This may give confusion to user.

Thus, letting users to create their own launcher may be clearer I think. What's your opinon @stonier?

stonier, 2015-04-10:
You can still load up the defaults yaml. Just let the user specify a custom parameter file which will do the overrides. That way they don't need to repeat the whole thing. We do this in `turtlebot_navigation/launch/move_base.xml`.

```
  <arg name="custom_param_file" default="$(find turtlebot_navigation)/param/dummy.yaml"/>

  <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
    <rosparam file="$(find turtlebot_navigation)/param/costmap_common_params.yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find turtlebot_navigation)/param/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find turtlebot_navigation)/param/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find turtlebot_navigation)/param/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find turtlebot_navigation)/param/dwa_local_planner_params.yaml" command="load" />
    <rosparam file="$(find turtlebot_navigation)/param/move_base_params.yaml" command="load" />
    <rosparam file="$(find turtlebot_navigation)/param/global_planner_params.yaml" command="load" />
    <rosparam file="$(find turtlebot_navigation)/param/navfn_global_planner_params.yaml" command="load" />
    <!-- external params file that could be loaded into the move_base namespace -->
    <rosparam file="$(arg custom_param_file)" command="load" />
```

Note all the default files that get loaded. And then one (explicitly `after` the others) that is the user's custom parameters that can add new params, or change ones that were set in the defaults.

stonier, 2015-04-10:
True.

How about stripping back the yaml file to just list up the pre-loaded advertisement/pull/flip rules? I think that is still useful without requiring a new launcher.

stonier, 2015-07-30:
Implemented a custom config override in 6fd34510f9f413fe6b73ece70b47deffe7753049.

stonier, 2015-07-30:
Cherry picked to develop branch in efba97f523fa64ce78b472bee00cb488408ed10b

closed 2015-07-30 by stonier