robotics-in-concert/rocon_multimaster #297: Cant run 'flip' script more than once

vikrantvm, 2015-04-24:
I am trying to send flip requests by running the flip script in rocon_gateway. The script runs successfully for the first time but fails when I try running it again:
 
```
Traceback (most recent call last):
  File "/home/vikrant/rocon/src/rocon_multimaster/rocon_gateway/scripts/flip", line 266, in <module>
    remote_rules = resolve_connections(gateway_namespace, remote_gateways, flipped_connections)
  File "/home/vikrant/rocon/src/rocon_multimaster/rocon_gateway/scripts/flip", line 185, in resolve_connections
    if not is_in_flipped_connection_list(remote_gateways, connection, flipped_connections)]
  File "/home/vikrant/rocon/src/rocon_multimaster/rocon_gateway/scripts/flip", line 168, in is_in_flipped_connection_list
    if flipped_connection.rule.name == connection.rule.name and \
AttributeError: 'RemoteRuleWithStatus' object has no attribute 'rule'
```
I also tried running the script on another machine and faced similar error. Im running the rocon gateway tutorial concert file. Inputs on how I could fix this error?

jihoonl, 2015-04-27:
Hello @vikrantvm, I have confirmed the bug. I will see how it can be fixed.

jihoonl, 2015-04-27:
There was a bug in the script. It should have been fixed.

closed 2015-04-27 by jihoonl

vikrantvm, 2015-04-27:
Thanks. I have a question, not sure if I should open another issue for that - but i'll just ask here. 
```How do I exchange parameters between different gateways? ```
I added a parameter to the remote gateway info msg and also in the script to display it(also loaded the param from the setup ros param script). But I just get an empty string for the new parameter, but all other params(firewall, hash name, etc) are displayed correctly. This works perfectly in the gateway info script and I get my new parameter by runnning the script.

jihoonl, 2015-04-27:
`rocon_gateway` does not support parameter flipping. It only supports services, topics, and actions.
You may need some extra scripts to send parameter info as topic or service to other master and set it as their param.

Also you can use [rosapi](http://wiki.ros.org/rosapi) to set rosparam. e.g)

Assume that you want to set rosparam from A to B.

1 ) flip /rosapi/set_param service server from master B to A
2)  call /rosapi/set_param service from master A
3) The param will set in B

vikrantvm, 2015-04-27:
Thanks for the information! I will try doing that...I'm not sure if the parameters are accessed(read-only) the way you mentioned above in the remote_gateway_info script? We can query the firewall/ip/name parameters of a remote gateway there.

commit 7b493040 (2015-04-27): fix the wrong message format parsing. closes #297