robotics-in-concert/rocon_multimaster #298: issue with PPP over serial link

dhirajdhule, 2015-05-26:
I am trying to use multimaster on two separate systems connected through PPP(no hardware flow control) over serial link (null modem cable). I can do ssh, netcat, etc. 
I followed gateway tutorials and I have encountered following problems.
1) (concert by zeroconf tutorial) When the remote gateway tries to connect to hub, zeroconf on remote gateway shows msg of service discovery timeout. is there anything we can do about it? like increasing the timeout.
2) (concert by param tutorial) The remote gateway when given IP of hub's system, was able to connect and flips were working but after approximately 30 sec the hub is showing error msg that the remote gateway is disconnected. This is happening repeatedly. I checked that there are no problems with communication link.

jihoonl, 2015-05-27:
Can you provide more detailed information? e.g) settings of ROS_MASTER_URI and ROS_HOSTNAME of both machines. logs in terminal...

stonier, 2015-08-18:
Closing due to lack of updates.

closed 2015-08-18 by stonier