robotics-in-concert/rocon_multimaster #299: Do not send flip request after adding flip rule in robot

dwlee, 2015-06-02:
I tried to do delivery service, repeatedly.
In normal case, robot showed following log .
```
[INFO] [WallTime: 1433157913.826239] Rapp Manager : dwleeab69e6702d26418eb6018116a9366941
[INFO] [WallTime: 1433157914.331153] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/map,subscriber)]
[INFO] [WallTime: 1433157914.331749] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/map_metadata,subscriber)]
[INFO] [WallTime: 1433157914.332177] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/annotation/tables,subscriber)]
[INFO] [WallTime: 1433157914.332540] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/annotation/ar_markers,subscriber)]
[INFO] [WallTime: 1433157914.332903] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/annotation/viz_tables,subscriber)]
[INFO] [WallTime: 1433157914.334867] Rapp Manager : successfully flipped ['map', 'map_metadata', 'tables', 'ar_markers', 'viz_tables']
[INFO] [WallTime: 1433157914.337892] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/young/robot_pose,publisher)]
[INFO] [WallTime: 1433157914.338520] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/young/robot_status,publisher)]
[INFO] [WallTime: 1433157914.340588] Rapp Manager : successfully flipped ['robot_pose', 'robot_status']
[INFO] [WallTime: 1433157914.344462] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/delivery_order_0771d9df6c4641f7a8c6c36556765ef6,action_server)]
[INFO] [WallTime: 1433157914.347544] Rapp Manager : successfully flipped ['delivery_order_0771d9df6c4641f7a8c6c36556765ef6']
[INFO] [WallTime: 1433157917.904285] Gateway : sending flip request [dwleeab69e6702d26418eb6018116a9366941][subscriber][/map][/young/amcl]
[INFO] [WallTime: 1433157918.123079] Gateway : sending flip request [dwleeab69e6702d26418eb6018116a9366941][subscriber][/annotation/ar_markers][/young/tracker_manager]
[INFO] [WallTime: 1433157918.261181] Gateway : sending flip request [dwleeab69e6702d26418eb6018116a9366941][subscriber][/annotation/ar_markers][/young/docking_interactor]
[INFO] [WallTime: 1433157918.405498] Gateway : sending flip request [dwleeab69e6702d26418eb6018116a9366941][publisher][/young/robot_status][/young/office_waiterbot_state_manager]
[INFO] [WallTime: 1433157918.532766] Gateway : sending flip request [dwleeab69e6702d26418eb6018116a9366941][publisher][/young/robot_pose][/young/robot_pose_publisher]

```
In above log, concert client sends flip request as soon as adding flip rule. 
However, it did not occasionally send flip request after adding flip rule like following log.

```
process[young/robot_pose_publisher-57]: started with pid [743]
[INFO] [WallTime: 1433158092.529331] Rapp Manager : dwleeab69e6702d26418eb6018116a9366941
[INFO] [WallTime: 1433158093.033828] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/map,subscriber)]
[INFO] [WallTime: 1433158093.034495] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/map_metadata,subscriber)]
[INFO] [WallTime: 1433158093.034905] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/annotation/tables,subscriber)]
[INFO] [WallTime: 1433158093.035358] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/annotation/ar_markers,subscriber)]
[INFO] [WallTime: 1433158093.035804] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/annotation/viz_tables,subscriber)]
[INFO] [WallTime: 1433158093.039067] Rapp Manager : successfully flipped ['map', 'map_metadata', 'tables', 'ar_markers', 'viz_tables']
[INFO] [WallTime: 1433158093.042436] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/young/robot_pose,publisher)]
[INFO] [WallTime: 1433158093.042919] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/young/robot_status,publisher)]
[INFO] [WallTime: 1433158093.044773] Rapp Manager : successfully flipped ['robot_pose', 'robot_status']
[INFO] [WallTime: 1433158093.048382] Gateway : added flip rule [dwleeab69e6702d26418eb6018116a9366941:(/delivery_order_6fb58c0fe30245e1bfbddb14ac58c077,action_server)]
[INFO] [WallTime: 1433158093.051253] Rapp Manager : successfully flipped ['delivery_order_6fb58c0fe30245e1bfbddb14ac58c077']

[ INFO] [1433158107.443481595]: Zeroconf: resolved service [xavier][_ros-multimaster-hub._tcp][local][3][ipv4][192.168.10.32:6380]

[INFO] [WallTime: 1433158116.938716] Gateway : sending unflip request [dwleeab69e6702d26418eb6018116a9366941][subscriber][/annotation/tables][/young/semantic_navigator]
[INFO] [WallTime: 1433158117.911817] Gateway : sending unflip request [dwleeab69e6702d26418eb6018116a9366941][action_server][/delivery_order_0771d9df6c4641f7a8c6c36556765ef6][/young/office_waiterbot_state_manager]
```
I will follow this issue and describe the more detail situation.

jihoonl, 2015-06-03:
There are the robot side logs.

We need to identify when `added flip rule`, `sending flip request` and `sending unflip request` are happening. 

* Find who adds flip rule when?
* When does sending flip requests happen?
* When does sending unflip requests happen? 
* What are the status of gateway?
* What is the status of concert side gateway and hub?

dwlee, 2015-06-08:
Waiterbot successes adding flip rule but does not send flip request.
In client log, gateway of client side continuously lost connection with hub.
 
client log (waiter bot)

```
NODES
  /young/camera/
    camera_nodelet_manager (nodelet/nodelet)
    driver (nodelet/nodelet)
    rectify_color (nodelet/nodelet)
  /young/
    amcl (amcl/amcl)
    ar_pair_tracking (yocs_ar_pair_tracking/yocs_ar_pair_tracking_node)
    ar_track_alvar (ar_track_alvar/individualMarkersNoKinect)
    depthimage_to_laserscan (nodelet/nodelet)
    dock_drive (nodelet/nodelet)
    docking_interactor (yocs_docking_interactor/docking_interactor_node)
    ir_link_publisher (tf/static_transform_publisher)
    ir_to_laserscan (waiterbot_sensors/waiter_ir_scan_node)
    localization_manager (yocs_localization_manager/localization_manager.py)
    move_base (move_base/move_base)
    navigation_velocity_smoother (nodelet/nodelet)
    office_waiterbot_state_manager (simple_delivery_robot_ctrl/waiterbot.py)
    reconfig_vel_mux (rosservice/rosservice)
    robot_pose_publisher (robot_pose_publisher/robot_pose_publisher)
    semantic_navigator (yocs_navigator/yocs_semantic_navigator_node)
    tracker_manager (yocs_localization_manager/tracker_manager.py)

ROS_MASTER_URI=http://localhost:11311
core service [/rosout] found
process[young/camera/camera_nodelet_manager-1]: started with pid [7945]
[ INFO] [1433724706.622775259]: Initializing nodelet with 4 worker threads.
process[young/camera/driver-2]: started with pid [7969]
[ INFO] [1433724706.997456079]: Loading nodelet /young/camera/driver of type openni2_camera/OpenNI2DriverNodelet to manager camera_nodelet_manager with the following remappings:
[ INFO] [1433724706.997595622]: /young/camera/delivery_order -> /delivery_order_b33097cd31ac4179918bc20cb55389b3
[ INFO] [1433724706.997651217]: /young/camera/depth/image -> /young/camera/depth_registered/image_raw
[ INFO] [1433724706.997697462]: /young/camera/map -> /map
[ INFO] [1433724706.997743312]: /young/camera/map_metadata -> /map_metadata
[ INFO] [1433724706.997784933]: /young/camera/marker_pose_list -> /annotation/ar_markers
[ INFO] [1433724706.997825511]: /young/camera/rgb/image -> /young/camera/rgb/image_raw
[ INFO] [1433724706.997869292]: /young/camera/table_pose_list -> /annotation/tables
[ INFO] [1433724706.997911743]: /young/camera/viz_marker_pose_list -> /annotation/viz_tables
[ INFO] [1433724707.129027519]: Device "1d27/0601@2/4" found.
Warning: USB events thread - failed to set priority. This might cause loss of data...
process[young/camera/rectify_color-3]: started with pid [8021]
[ INFO] [1433724707.494876721]: Loading nodelet /young/camera/rectify_color of type image_proc/rectify to manager camera_nodelet_manager with the following remappings:
[ INFO] [1433724707.495019222]: /young/camera/delivery_order -> /delivery_order_b33097cd31ac4179918bc20cb55389b3
[ INFO] [1433724707.495199145]: /young/camera/image_mono -> /young/camera/rgb/image_raw
[ INFO] [1433724707.495298309]: /young/camera/image_rect -> /young/camera/rgb/image_rect_color
[ INFO] [1433724707.495388446]: /young/camera/map -> /map
[ INFO] [1433724707.495473740]: /young/camera/map_metadata -> /map_metadata
[ INFO] [1433724707.495559158]: /young/camera/marker_pose_list -> /annotation/ar_markers
[ INFO] [1433724707.495641072]: /young/camera/table_pose_list -> /annotation/tables
[ INFO] [1433724707.495723786]: /young/camera/viz_marker_pose_list -> /annotation/viz_tables
process[young/depthimage_to_laserscan-4]: started with pid [8078]
[ INFO] [1433724707.930124729]: Loading nodelet /young/depthimage_to_laserscan of type depthimage_to_laserscan/DepthImageToLaserScanNodelet to manager camera/camera_nodelet_manager with the following remappings:
[ INFO] [1433724707.930272843]: /young/camera/image -> /young/camera/depth/image_raw
[ INFO] [1433724707.930326515]: /young/camera/scan -> /young/scan
[ INFO] [1433724707.930411955]: /young/delivery_order -> /delivery_order_b33097cd31ac4179918bc20cb55389b3
[ INFO] [1433724707.930489331]: /young/image -> /young/camera/depth/image_raw
[ INFO] [1433724707.930558760]: /young/map -> /map
[ INFO] [1433724707.930651838]: /young/map_metadata -> /map_metadata
[ INFO] [1433724707.930722901]: /young/marker_pose_list -> /annotation/ar_markers
[ INFO] [1433724707.930790747]: /young/table_pose_list -> /annotation/tables
[ INFO] [1433724707.930859433]: /young/viz_marker_pose_list -> /annotation/viz_tables
process[young/ir_link_publisher-5]: started with pid [8133]
process[young/ir_to_laserscan-6]: started with pid [8164]
[ INFO] [1433724708.601589125]: IR Scan node : initialized
[ INFO] [1433724708.601757618]: uniserial::initialize(): Initializing on defined settings.
[... 172 lines cut ...]
  Pruned Tree
    rocon:/waiterbot|turtlebot|pc : ['young']
[INFO] [WallTime: 1433724698.330978] Scheduler : compatibility tree is valid, attempting to allocate [d1b129a2-7466-436e-bd0d-f11676224e2a]

...

[WARN] [WallTime: 1433724726.838612] Hub Watcher: gateway young517847829ff94062914e80113b819660 has been unavailable for 60 seconds! Marking as unavailable.

...

[INFO] [WallTime: 1433724727.489227] Conductor : concert client transition [available->missing][young]
[ INFO] [1433724727.576539232]: Remove subscriber of : 

...

[INFO] [WallTime: 1433724766.714433] Conductor : concert client transition [missing->available][young]
[ INFO] [1433724766.776599461]: Create Subscriber for : young	Topic : /young/robot_pose

...

[WARN] [WallTime: 1433725022.055619] Hub Watcher: gateway young517847829ff94062914e80113b819660 has been unavailable for 60 seconds! Marking as unavailable.
[INFO] [WallTime: 1433725022.268569] Conductor : concert client transition [available->missing][young]
[ INFO] [1433725022.276596831]: Remove subscriber of : 

```

jihoonl, 2015-06-08:
It is expected behavior. `Sending flip request` cannot happen if hub is invisible.

#301 and e81c419 should resolve this issue as they prevents frequent hub loss.

dwlee, 2015-06-09:
I found the more detail reason regarding this issue, so I reopen this issue. 

In normal case, all flip rules are removed and then gateway sends unflipping request like following log when rapp is stopped. 
It is no problem to start `start_rapp` service again in this case.
```
[INFO] [WallTime: 1433817015.812745] Rapp Manager : stopped rapp [waiterbot_rapps/pickup_delivery]'.
[INFO] [WallTime: 1433817015.818353] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/map,subscriber)]
[INFO] [WallTime: 1433817015.819460] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/map_metadata,subscriber)]
[INFO] [WallTime: 1433817015.820309] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/annotation/tables,subscriber)]
[INFO] [WallTime: 1433817015.821167] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/annotation/ar_markers,subscriber)]
[INFO] [WallTime: 1433817015.821909] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/annotation/viz_tables,subscriber)]
[INFO] [WallTime: 1433817015.825246] Rapp Manager : successfully flipped ['map', 'map_metadata', 'tables', 'ar_markers', 'viz_tables']
[INFO] [WallTime: 1433817015.830399] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/young/robot_pose,publisher)]
[INFO] [WallTime: 1433817015.831374] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/young/robot_status,publisher)]
[INFO] [WallTime: 1433817015.834550] Rapp Manager : successfully flipped ['robot_pose', 'robot_status']
[INFO] [WallTime: 1433817015.839540] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/delivery_order_d5f55789ed3648cc87fd5a77b5897c87,action_server)]
[INFO] [WallTime: 1433817015.842278] Rapp Manager : successfully flipped ['delivery_order_d5f55789ed3648cc87fd5a77b5897c87']
[INFO] [WallTime: 1433817031.091992] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][subscriber][/annotation/tables][/young/semantic_navigator]
[INFO] [WallTime: 1433817033.909518] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][subscriber][/annotation/ar_markers][/young/tracker_manager]
[INFO] [WallTime: 1433817034.974858] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][subscriber][/annotation/ar_markers][/young/docking_interactor]
[INFO] [WallTime: 1433817035.851742] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][subscriber][/map][/young/amcl]
[INFO] [WallTime: 1433817037.002432] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][publisher][/young/robot_status][/young/office_waiterbot_state_manager]
[INFO] [WallTime: 1433817037.833788] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][publisher][/young/robot_pose][/young/robot_pose_publisher]
[INFO] [WallTime: 1433817038.497595] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][action_server][/delivery_order_d5f55789ed3648cc87fd5a77b5897c87][/young/office_waiterbot_state_manager]
```

However, the problem situation is happened when `start_rapp` service is called while removing flip rule and sending unflip request like following log.

```
[INFO] [WallTime: 1433817639.977617] Rapp Manager : stopped rapp [waiterbot_rapps/pickup_delivery]'.
[INFO] [WallTime: 1433817639.983239] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/map,subscriber)]
[INFO] [WallTime: 1433817639.984393] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/map_metadata,subscriber)]
[INFO] [WallTime: 1433817639.985267] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/annotation/tables,subscriber)]
[INFO] [WallTime: 1433817639.986058] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/annotation/ar_markers,subscriber)]
[INFO] [WallTime: 1433817639.986860] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/annotation/viz_tables,subscriber)]
[INFO] [WallTime: 1433817639.990089] Rapp Manager : successfully flipped ['map', 'map_metadata', 'tables', 'ar_markers', 'viz_tables']
[INFO] [WallTime: 1433817639.995199] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/young/robot_pose,publisher)]
[INFO] [WallTime: 1433817639.996181] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/young/robot_status,publisher)]
[INFO] [WallTime: 1433817639.999124] Rapp Manager : successfully flipped ['robot_pose', 'robot_status']
[INFO] [WallTime: 1433817640.003701] Gateway : removed flip rule [dwlee8976b74a4d054bd59c436ddeeed2a3bc:(/delivery_order_f8f2689400d94eb79ac10d34d39cd13f,action_server)]
[INFO] [WallTime: 1433817640.006226] Rapp Manager : successfully flipped ['delivery_order_f8f2689400d94eb79ac10d34d39cd13f']
[INFO] [WallTime: 1433817642.166565] Rapp Manager : request received to start rapp [concert_common_rapps/pickup_delivery]
concert_common_rapps/pickup_delivery
[INFO] [WallTime: 1433817642.167473] Rapp Manager : starting app 'concert_common_rapps/pickup_delivery' underneath young
[WARN] [WallTime: 1433817648.438255] Rapp Manager : ignoring uncessary remapping rule (ir, ir)
[WARN] [WallTime: 1433817648.438725] Rapp Manager : ignoring uncessary remapping rule (rgb, rgb)
[WARN] [WallTime: 1433817648.439124] Rapp Manager : ignoring uncessary remapping rule (depth, depth)

...

process[young/robot_pose_publisher-513]: started with pid [3728]
[... cut ...]
[INFO] [WallTime: 1433817688.505595] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][subscriber][/annotation/tables][/young/semantic_navigator]
...
[INFO] [WallTime: 1433817696.766228] Gateway : sending unflip request [dwlee8976b74a4d054bd59c436ddeeed2a3bc][action_server][/delivery_order_f8f2689400d94eb79ac10d34d39cd13f][/young/office_waiterbot_state_manager]
...

```

I think that the topics (ex. /map, /map_metadata,etc ) can not be flipped if the flip rules regarding topics is added before removing and unflipping.

jihoonl, 2015-06-11:
**Summary**

* Rapp manager is continuously starting and stopping the same rapp.(e.g navigation rapp).
* For Every rapp stop/start requests, **"/move_base" node with different xmlrpc uri** starts up and stops
* Rapp manager requests to flip "/map" subscriber for every rapp starting, and requests unflip for rapp stopping.
* When Rapp stopping and restarting is faster then gateway watch loop period, `Removing flip rule` and `adding flip rule` does not get detected by watcher thread. 
* Thus it does not create any new flip request. Though it supposed to unflip old "/map" subscriber, and flip new "/map" subscriber with new "move_base" xmlrpc uri.

jihoonl, 2015-06-12:
It seems like the easiest way to fix this issue is add [xmlrpc](https://github.com/robotics-in-concert/rocon_multimaster/blob/indigo/rocon_gateway/src/rocon_gateway/flipped_interface.py#L263-L267) in flip_rule list. Then it will automatically remove old nodes flip rules and add new nodes flip rule.

PR #309 (2015-06-17): Refactor gateway logics
It fixes #299.

Tested and working fine.
closed 2015-06-17 by jihoonl