robotics-in-concert/rocon_multimaster #301: Can clients get message via flipped topic when gateway losts connection with hub?

dwlee, 2015-06-02:
I was wondering that client can get message following situation. More detail, can this client get map data via "/map" topic?

```
[INFO] [WallTime: 1433209010.488213] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][action_server][/delivery_order_cebf99134db8431bad2fc276d1619b5f][/young/waiterbot_state_manager]
[INFO] [WallTime: 1433209045.666824] Gateway : lost connection with hub, attempting to disconnect...
[INFO] [WallTime: 1433209045.698764] Gateway : disengaged connection with the hub [xavier][192.168.10.32:6380]
[INFO] [WallTime: 1433209047.082658] Gateway : unresolvable hub [couldn't connect to the redis server]
[INFO] [WallTime: 1433209053.459145] Gateway : Found existing mismatched public key on the hub. Requesting resend for all flip-ins.
[INFO] [WallTime: 1433209055.677703] Gateway : registering on the hub [xavier]
[INFO] [WallTime: 1433209055.679687] Gateway : discovered hub via zeroconf [192.168.10.32:6380]
[INFO] [WallTime: 1433209056.683423] Gateway : unresolvable hub [couldn't connect to the redis server]
[INFO] [WallTime: 1433209061.070734] Gateway : blacklisting hub [already connected to this hub]
[INFO] [WallTime: 1433209061.212334] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][subscriber][/annotation/tables][/young/semantic_navigator]
[INFO] [WallTime: 1433209066.231486] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][subscriber][/vm_delivery/drink_order_result][/young/vm_interactor]
[INFO] [WallTime: 1433209067.920826] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][subscriber][/annotation/ar_markers][/young/tracker_manager]
[INFO] [WallTime: 1433209068.938569] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][subscriber][/annotation/ar_markers][/young/docking_interactor]
[INFO] [WallTime: 1433209071.992204] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][subscriber][/map][/young/amcl]
[INFO] [WallTime: 1433209073.022550] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][publisher][/vm_delivery/drink_order][/young/vm_interactor]
[INFO] [WallTime: 1433209074.994853] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][publisher][/young/robot_pose][/young/robot_pose_publisher]
[INFO] [WallTime: 1433209077.459363] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][publisher][/young/robot_status][/young/waiterbot_state_manager]
[INFO] [WallTime: 1433209078.063074] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][service][/young33862b93e6a64e4e8995442ef1542d49/start_rapp][/app_manager]
[INFO] [WallTime: 1433209079.324676] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][service][/young33862b93e6a64e4e8995442ef1542d49/stop_rapp][/app_manager]
[INFO] [WallTime: 1433209080.761006] Gateway : sending flip request [xavier3156d285be33488b8275844308020367][action_server][/delivery_order_cebf99134db8431bad2fc276d1619b5f][/young/waiterbot_state_manager]
[ WARN] [1433209140.582185791]: Zeroconf: timed out resolving service [xavier][_ros-multimaster-hub._tcp][local][3][ipv4][192.168.10.32:6380]
```

jihoonl, 2015-06-03:
Please add log from concert master side.

jihoonl, 2015-06-03:
Also the following symptom seems to happen a lot even if robot is in the stable network. Should check when it loses connection with hub. 

```
[INFO] [WallTime: 1433209045.666824] Gateway : lost connection with hub, attempting to disconnect...
[INFO] [WallTime: 1433209045.698764] Gateway : disengaged connection with the hub [xavier][192.168.10.32:6380]
[INFO] [WallTime: 1433209047.082658] Gateway : unresolvable hub [couldn't connect to the redis server]
[INFO] [WallTime: 1433209053.459145] Gateway : Found existing mismatched public key on the hub. Requesting resend for all flip-ins.
[INFO] [WallTime: 1433209055.677703] Gateway : registering on the hub [xavier]
[INFO] [WallTime: 1433209055.679687] Gateway : discovered hub via zeroconf [192.168.10.32:6380]
[INFO] [WallTime: 1433209056.683423] Gateway : unresolvable hub [couldn't connect to the redis server]
[INFO] [WallTime: 1433209061.070734] Gateway : blacklisting hub [already connected to this hub]
```

jihoonl, 2015-06-03:
lost connection with hub problem was because of [too short socket timeout in hub pinging logic](https://github.com/robotics-in-concert/rocon_multimaster/blob/indigo/rocon_hub_client/src/rocon_hub_client/hub_client.py#L26-L62). HubConnection instance has 1.0sec socket timeout and it causes frequent hub disengagement. (Actually, gateway cannot be used for network infra that has more than 1 sec latency because of this logic).

Increase this to hub ping rate would resolve the problem

commit e81c419b (2015-06-03): increase hubconnection socket timeout to resolve frequent hub disengagement closes #301
closed 2015-06-03 by jihoonl