robotics-in-concert/rocon_multimaster #305: parameterise  hub ping frequency

jihoonl, 2015-06-09:
https://github.com/robotics-in-concert/rocon_multimaster/blob/indigo/rocon_gateway/src/rocon_gateway/gateway_hub.py#L42

This needs to do. It should get applied to 

* https://github.com/robotics-in-concert/rocon_multimaster/blob/indigo/rocon_gateway/src/rocon_gateway/gateway_hub.py#L59
* https://github.com/robotics-in-concert/rocon_multimaster/blob/indigo/rocon_hub_client/src/rocon_hub_client/hub_client.py#L92
* https://github.com/robotics-in-concert/rocon_multimaster/blob/indigo/rocon_hub_client/src/rocon_hub_client/hub_discovery.py#L175

stonier, 2015-08-18:
We could, though I'm inclined not to unless there is a need (proliferation of parameters).

Also note there are two loops above, and probably should remain separate entities:

1. The hub discovery loop
2. The hub pinger (disconnection callback) loop

stonier, 2015-08-18:
Reopen if there is a need for this.

closed 2015-08-18 by stonier