robotics-in-concert/rocon_multimaster #306: Redis v 3.0.1

asmodehn, 2015-06-12:
[INFO] [WallTime: 1434073128.708689] Hub : version 3.0.1
[FATAL] Hub : the version of the redis server you have installed is not supported by rocon.
[FATAL] Hub : please submit a ticket at https://github.com/robotics-in-concert/rocon_multimaster

I wonder what is involved in supporting different version of redis in rocon_multimaster ?

jihoonl, 2015-06-12:
These would be related. All OS support different versions of redis.

https://github.com/robotics-in-concert/rocon_multimaster/issues/247
https://github.com/robotics-in-concert/rocon_multimaster/issues/277

jihoonl, 2015-07-15:
Is this fixed by #313? #313 added config for redis 3.0

stonier, 2015-08-18:
Removed the experimental config files (8d6cd0b5) as it's misleading while we can't officially claim support. Can always pull them back later when we wish to try.

Closing.

closed 2015-08-18 by stonier