robotics-in-concert/rocon_multimaster #314: Issues found while testing network with GoCart Concert setup

asmodehn, 2015-07-13:
* [ ] dum de dooh
* [x] dum de dah

jihoonl, 2015-07-13:
#310 also fails in [register_gateway](https://github.com/robotics-in-concert/rocon_multimaster/blob/develop/rocon_gateway/src/rocon_gateway/gateway_hub.py#L98-L147).

Would be similar issue.

stonier, 2015-07-14:
Q) Happening every time?

asmodehn, 2015-08-04:
* [] #317 Conductor confused if pull rules already exists -> this applies to both 'missing' and 'gone' robot states.
* [] #318 robot gateway not cleaning it s flipped topics on shutdown -> this is a nice to have, but should not be relied on to guaranteed the system is always working.

stonier, 2015-08-18:
Closing this. Fixed some issues with the flipping (but not pulling). Conductor-gateway connectivity should use a more systematic approach to providing data and reproducing bugs, finding root causes and making plans to fix than the noise we currently have.

Let's close this and tackle the problem anew with the gopher conductor.

closed 2015-08-18 by stonier