robotics-in-concert/rocon_multimaster #318: Robot Gateway not cleaning its flipped topics on shutdown

asmodehn, 2015-08-04:
Setup : 
- Concert running
- Robot running
- We kill the robot.
- Then on the concert we check the gateway information : 
```
alexv@alexv-PC:~/ROS/gopher_concert$ gateway_info 
Gateway
  Namespace   : /concert/gateway
  Name        : GoCart Concert
  Hash Name   : GoCart Concert
  Namespace   : /concert/gateway
  Ip/Hostname : localhost
  Connected   : True
  Hubs        : GoCart Concert [localhost:6380]
  Firewall    : False
  Public Watchlist
       : -
  Public Interface
       : -
  Flip Watchlist
       : -
  Flipped Connections
       : -
  Flipped in Connections
       : -
  Pull Watchlist
       : tosh-/tosh/status-publisher-.*
       : tosh-/tosh/invite-service-.*
  Pulled Connections
       : -
```
It seems the pull watchlist here should be empty.
But since this was set by the robot gateway, the robot gateway should clean it up.

Note : This is lower priority than #317.
Form the point of view of the surviving gateway, you can never be sure if the disappeared gateway is dead, or just missing , or had a bug in it, or just happened to have a strange behavior on that revision at the customer site.
=> the gateway should not rely on this cleaning up happening. It s just a nice thing to have to avoid an ever-increasing redis DB, but this behavior cannot be guaranteed and should not be relied on.

stonier, 2015-08-18:
> But since this was set by the robot gateway, the robot gateway should clean it up.

This was not set by the robot gateway.

The concert is doing the pulling, and hence the concert gateway that is setting those pull rules to be watched for. 

Not enough data here to determine if advertisements from the robot gateway were actually left lying around and confusing the conductor.

stonier, 2015-08-18:
Fixed two bugs in the gateway flipping process in 216823f4 and since the conductor workflow is now different, I'll close this.

If we get more relevant data re the gateways, make a new issue.

closed 2015-08-18 by stonier