robotics-in-concert/rocon_multimaster #330: Post Connection Cache Bugs

stonier, 2016-01-26:
** Pulls aren't cancelled **

```
0> rocon_launch rocon_gateway_tutorials gateway_tutorials.concert --screen
2> rosrun rocon_gateway_tutorials advertise_all.py
3> rosrun rocon_gateway_tutorials pull_all.py
3> rosrun rocon_gateway_tutorials pull_all.py --cancel
2> gateway_info (or rocon_gateway_graph)
```

All the connections are cancelled at the 3rd ros master, but you can still see the pulls registered there, which means some information is still left lying around the server.

asmodehn, 2016-01-29:
Couldn't really reproduce, I think : 

gateway_info on 11312 : 
```
Gateway
  Namespace   : /gateway
  Name        : pirate_gateway
  Hash Name   : pirate_gateway26b31b1563e849c585be71dd33df8b58
  Namespace   : /gateway
  Ip/Hostname : asmodehn
  Connected   : True
  Hubs        : Pirate Hub [localhost:6380]
  Firewall    : False
  Public Watchlist
       : .*-subscriber-.*
       : .*-publisher-.*
       : .*-action_client-.*
       : .*-service-.*
       : .*-action_server-.*
  Public Interface
       : /chatter-subscriber-/listener
       : /chatter-publisher-/talker
       : /babbler-publisher-/babbler
       : /fibonacci/client-action_client-/fibonacci/client/fibonacci_client
       : /add_two_ints-service-/add_two_ints_server
       : /fibonacci/server-action_server-/fibonacci/server/fibonacci_server
  Flip Watchlist
       : -
  Flipped Connections
       : -
  Flipped in Connections
       : -
  Pull Watchlist
       : -
  Pulled Connections
       : -
```
These are the advertisements I think, correct ?

gateway_info on 11313 : 
```
Gateway
  Namespace   : /gateway
  Name        : pirate_gateway
  Hash Name   : pirate_gateway84b2ffff66eb40c589be3b9c574da1b8
  Namespace   : /gateway
  Ip/Hostname : asmodehn
  Connected   : True
  Hubs        : Pirate Hub [localhost:6380]
  Firewall    : False
  Public Watchlist
       : -
  Public Interface
       : -
  Flip Watchlist
       : -
  Flipped Connections
       : -
  Flipped in Connections
       : -
  Pull Watchlist
       : -
  Pulled Connections
       : -

```

asmodehn, 2016-01-29:
I just put together very small fixes there : https://github.com/robotics-in-concert/rocon_multimaster/pull/331
Maybe part of it fixes some issue here...

stonier, 2016-01-29:
Looks like pull cancelling is cancelling, but now throwing a warning:

```
[WARN] [WallTime: 1454064753.846667] Trying to access nodes for inexistent /fibonacci/client/cancel in {'/zeroconf/new_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ccd0>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf50>, '/babbler': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ced0>, '/rosout': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf90>, '/connection_cache/list': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c2d0>, '/rosout_agg': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce10>, '/zeroconf/lost_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cad0>, '/gateway/gateway_info': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cbd0>, '/connection_cache/diff': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c6d0>, '/chatter': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c1d0>} 
[WARN] [WallTime: 1454064753.847143] Trying to access nodes for inexistent /fibonacci/server/status in {'/zeroconf/new_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ccd0>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf50>, '/babbler': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ced0>, '/rosout': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf90>, '/connection_cache/list': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c2d0>, '/rosout_agg': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce10>, '/zeroconf/lost_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cad0>, '/gateway/gateway_info': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cbd0>, '/connection_cache/diff': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c6d0>} 
[WARN] [WallTime: 1454064753.847539] Trying to access nodes for inexistent /fibonacci/server/result in {'/zeroconf/new_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ccd0>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf50>, '/rosout': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf90>, '/connection_cache/list': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c2d0>, '/rosout_agg': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce10>, '/zeroconf/lost_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cad0>, '/gateway/gateway_info': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cbd0>, '/connection_cache/diff': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c6d0>} 
[WARN] [WallTime: 1454064753.847905] Trying to access nodes for inexistent /fibonacci/server/feedback in {'/zeroconf/new_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ccd0>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf50>, '/rosout': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf90>, '/connection_cache/list': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c2d0>, '/rosout_agg': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce10>, '/zeroconf/lost_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cad0>, '/gateway/gateway_info': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cbd0>, '/connection_cache/diff': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c6d0>} 
[WARN] [WallTime: 1454064753.848256] Trying to access nodes for inexistent /fibonacci/client/goal in {'/zeroconf/new_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ccd0>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf50>, '/rosout': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cf90>, '/connection_cache/list': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c2d0>, '/rosout_agg': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce10>, '/zeroconf/lost_connections': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cad0>, '/gateway/gateway_info': <rocon_python_comms.connections.Channel object at 0x7f4da0e1cbd0>, '/connection_cache/diff': <rocon_python_comms.connections.Channel object at 0x7f4da0e1c6d0>} 
[WARN] [WallTime: 1454064753.848570] Trying to access nodes for inexistent /fibonacci/client/status in {'/connection_cache/diff': <rocon_python_comms.connections.Channel object at 0x7f4da0dadd50>, '/chatter': <rocon_python_comms.connections.Channel object at 0x7f4da0dadd90>, '/rosout': <rocon_python_comms.connections.Channel object at 0x7f4da0dadd10>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce50>} 
[WARN] [WallTime: 1454064753.848860] Trying to access nodes for inexistent /fibonacci/client/result in {'/connection_cache/diff': <rocon_python_comms.connections.Channel object at 0x7f4da0dadd50>, '/chatter': <rocon_python_comms.connections.Channel object at 0x7f4da0dadd90>, '/rosout': <rocon_python_comms.connections.Channel object at 0x7f4da0dadd10>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce50>} 
[... cut ...] .Channel object at 0x7f4da0dadd10>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce50>} 
[WARN] [WallTime: 1454064753.849917] Trying to access nodes for inexistent /fibonacci/server/goal in {'/connection_cache/diff': <rocon_python_comms.connections.Channel object at 0x7f4da0dadd50>, '/rosout': <rocon_python_comms.connections.Channel object at 0x7f4da0dadd10>, '/connection_cache/spin': <rocon_python_comms.connections.Channel object at 0x7f4da0e1ce50>} 
```

stonier, 2016-01-29:
Also, flip cancelling is not working for actions - they get left lying around.

asmodehn, 2016-02-01:
These should address the relevant issues : 
https://github.com/robotics-in-concert/rocon_multimaster/pull/333
https://github.com/robotics-in-concert/rocon_tools/pull/122

One problem still remain though : 
```
$ rocon_launch rocon_gateway_tutorials gateway_tutorials.concert --screen
$ . masteruri 2
http://localhost:11312
$ rosrun rocon_gateway_tutorials flip_all.py
[INFO] [WallTime: 1454317851.457685] Flip All : flipping all [pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a].
$ rosrun rocon_gateway_tutorials flip_all.py --cancel
[INFO] [WallTime: 1454317858.819137] Flip All : cancelling all [pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a].
$ gateway_info 
Gateway
  Namespace   : /gateway
  Name        : pirate_gateway
  Hash Name   : pirate_gateway179637b5639f48d8a33cc7d6dfed8e04
  Namespace   : /gateway
  Ip/Hostname : asmodehn
  Connected   : True
  Hubs        : Pirate Hub [localhost:6380]
  Firewall    : False
  Public Watchlist
       : -
  Public Interface
       : -
  Flip Watchlist
       : -
  Flipped Connections
       : -
  Flipped in Connections
       : -
  Pull Watchlist
       : -
  Pulled Connections
       : -
$ remote_gateway_info 
Local Gateway
  Namespace: /gateway
  Ip/Hostname: asmodehn
Remote Gateway
  Name        : pirate_gateway
  Hash Name   : pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a
  Ip/Hostname : asmodehn
  Firewall    : false
  Alive       : True
  Connection  : unknown
  Latency from Hub(ms)
    Min/Max        : 0.000-0.000
    Average        : 0.000
    Mean Deviation : 0.000
  Public Interface
          : -
  Flipped Interface
          : -
  Pulled Interface
          : -
```
Everything looks fine. But :
```
$ . masteruri 3
http://localhost:11313
$ gateway_info 
Gateway
  Namespace   : /gateway
  Name        : pirate_gateway
  Hash Name   : pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a
  Namespace   : /gateway
  Ip/Hostname : asmodehn
  Connected   : True
  Hubs        : Pirate Hub [localhost:6380]
  Firewall    : False
  Public Watchlist
       : -
  Public Interface
       : -
  Flip Watchlist
       : -
  Flipped Connections
       : -
  Flipped in Connections
       : -
  Pull Watchlist
       : -
  Pulled Connections
       : -
$ remote_gateway_info 
Local Gateway
  Namespace: /gateway
  Ip/Hostname: asmodehn
Remote Gateway
  Name        : pirate_gateway
  Hash Name   : pirate_gateway179637b5639f48d8a33cc7d6dfed8e04
  Ip/Hostname : asmodehn
  Firewall    : false
  Alive       : True
  Connection  : unknown
  Latency from Hub(ms)
    Min/Max        : 0.000-0.000
    Average        : 0.000
    Mean Deviation : 0.000
  Public Interface
          : -
  Flipped Interface
          : pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a-/fibonacci/server-action_server-/fibonacci/server/fibonacci_server,http://asmodehn:45940/
          : pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a-/fibonacci/client-action_client-/fibonacci/client/fibonacci_client,http://asmodehn:53713/
          : pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a-/chatter-publisher-/talker,http://asmodehn:51041/
          : pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a-/chatter-subscriber-/listener,http://asmodehn:60961/
          : pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a-/add_two_ints-service-/add_two_ints_server,http://asmodehn:41143/
          : pirate_gatewayeb3c9543ac85479e917c0a9ca0be957a-/babbler-publisher-/babbler,http://asmodehn:33891/
  Pulled Interface
          : -

asmodehn, 2016-02-01:
got that last one. second commit in there https://github.com/robotics-in-concert/rocon_multimaster/pull/333
@stonier We re good to go I think.

stonier, 2016-02-03:
Everything now working after #333 and [rocon_tools/pull-requests/122](https://github.com/robotics-in-concert/rocon_tools/pull/122)

closed 2016-02-03 by stonier