robotics-in-concert/rocon_multimaster #336: invalid ROS names

asmodehn, 2016-03-21:
Need to be careful : as per specification [ROS name format](http://wiki.ros.org/Names) != [hostname format](https://en.wikipedia.org/wiki/Hostname).

Currently rocon uses hostname as robot name, but this should be properly formatted before.
We should probably have warning in multiple places to check for validity of ros names...

The effect of using an invalid ROS name are still not clear.
A warning message will be displayed by ROS : 
```
UserWarning: '/concert/clients/alexv-robotsim/battery' is not a legal ROS graph resource name. This may cause problems with other ROS tools
  super(Subscriber, self).__init__(name, data_class, Registration.SUB)
```

asmodehn, 2016-03-21:
Note : I was able to flip topics that contained a dash '-', so this warning should not prevent from flipping topics.

stonier, 2016-03-31:
Dashed topic will ultimately be unusable,  regardless

>
> Note : I was able to flip topics that contained a dash '-', so this
warning should not prevent from flipping topics.

stonier, 2017-01-26:
Closing this, the ros warning message is enough.

closed 2017-01-26 by stonier