robotics-in-concert/rocon_multimaster #348: Lunar Release?

mikaelarguedas, 2017-08-05:
Hi rocon maintainers!

Is there a plan to release rocon_multimaster and all the great robot in concert tools into ROS Lunar ?

stonier, 2017-08-06:
Wasn't planning on it (only using the long term releases). Is there a need for it? Can do if that is the case.

mikaelarguedas, 2017-08-06:
Thanks for the quick answer. There is no specific need right now I was just going through the top of the list of [blocked releases](http://repositories.ros.org/status_page/blocked_releases_lunar.html?s=6&r=1) in Lunar and so rocon in the top 3 hence the question. It's totally fine to not release it except if there there is a specific request for the community.

mikaelarguedas, 2019-08-04:
Lunar is EOL :wave:

closed 2019-08-04 by mikaelarguedas